BaptisteHudyma/Lamp-Da #114: Add user input action system

BaptisteHudyma, 2025-06-23:
Add a system to register user events on inputs : 
- button clicks
- button click and hold
- lamp tilt/shake

Modes can register on those events to make their own behaviors

deuxksix, 2025-07-31:
Custom clicks & click+hold behavior was the intended behavior behind `custom_hold` and `hasButtonCustomUI`

However, we need to decide how can users enable / disable that custom UI for a given mode — the WIP proposal was that users would be required to turn off the lamp, then turn it on again while holding the button / doing a 3H / something else.

Also, if you add a `imu_shake_default` and a `imu_tilt_default` to `indexable_behavior.hpp` I'll happily add all the required callbacks in the manager / modes for them to implement whatever they want with it :)

deuxksix, 2025-08-19:
These two (#195 #196) should do the trick. Added that to [1.1 milestone](https://github.com/BaptisteHudyma/Lamp-Da/milestone/1) as it would be a nice addition.