BaptisteHudyma/Lamp-Da #150: Migrate away from "legacy modes"

deuxksix, 2025-07-31:
Legacy modes directly use the "strip" object which prevent us to abstract away the hardware. An on-going goal of mine is to use `LampTy` as an unified interface to the hardware for the user, independently of which kind of mode they want to write.

Removing "direct strip access" from modes is also the required path to implement things like ramp animations, screen overlays, drawing functions with proper dithering, etc.

However, as most modes are for now still doing the `.getLegacyStrip()` direct access, new users are replicating this while writing new modes, and we're going to meme ourselves into a corner that will later break our users code & their custom modes.

To avoid that, we should take some time to migrate each legacy mode to its own file in `modes/default/<mode-name>.hpp` when appropriate (other modes accessing the hardware directly, as they do heavy access C-like hardware APIs, are less of priority and should wait for proper support in `LampTy` of their required feature, before migration)

--------------

This issue targets the "simple to migrate" legacy modes, another issue should follow for the IMU / sound FFT stuff.

deuxksix, 2025-08-19:
Breaking down which modes may be already started:

- [ ] `PolarMode` is ready to be migrated 💚  
   - 🟢 has `modes::colors::from_palette` as drop-in replacement for `get_color_from_palette`
   - 💡 add `modes::utils::linear_scale` (with documentation) as `lmpd_map` with _static_ boundaries
- [ ] `SineMode` is ready to be migrated 💚 
   - 🟢 has `modes::colors::from_palette` as drop-in replacement for `get_color_from_palette`
   - 💡 add `modes::colors::blend(leftColor, rightColor, blendAmount)` as `utils::color_blend` +simplify `b16`
   - ⚠️ using `colors::from_blend` to black may be replaced by `colors::fade`
- [ ] `DriftMode` is ready to be migrated 💚 
   - 🟢 has `modes::colors::from_palette` as drop-in replacement for `get_color_from_palette`
   - 💡 add `modes::colors::add<useFastAdd>(leftColor, rightColor)` as `utils::color_add` +`LMBD_INLINE`
   - 💡 add `modes::colors::fade<isVideoMode>(inputColor, fadeAmount)` as `utils::color_fade` +`LMBD_INLINE`
   - 💡 add `modes::hardware::LampTy::blur(blueAmount)`
   - ⚠️ may be able to use `ctx.lamp.tick` to replace the `time_ms() / 32` usage
- [ ] `DistMode` is ready to be migrated 💚 
   - ⚠️ maybe `utils::gamma8` can be migrated in a meaningful way
- [ ] `VuMeterMode` is ready to be migrated 💚 
  - 🎉 likely you can have fun having a ramp to configure how "reactive" meter is
  - 💡 may also add a new `VuMeterTy` inside `modes::audio` to help build beautiful sound reactive modes
  - ⚠️ shall replace `GenerateGradientColor` with `modes::colors::PaletteGradient`+`modes::colors::from_palette`
- [ ] `RainbowSwirlMode` not ready ❌
  - 🟢 has `modes::colors::fromAngleHue` to replace `hue_to_rgb_sinus`
  - 🟡  WIP decide how to migrate `Color`+`animations::fill`
- [ ] `PartyFadeMode` not ready ❌
   - 🟢 has `modes::colors::from_palette` as drop-in replacement for `get_color_from_palette`
   - 🟡 WIP decide how to migrate `Color`+`animations::fade_in`
   - 💡 add `modes::colors::from_gradient(startColor, endColor, position)` to replace `utils::get_gradient`
   - ⚠️ probably may drop `GeneratePaletteIndexed` which only wraps `get_color_from_palette`
   - ⚠️ probably may be simplified in its behavior
- [ ] `ColorWipeMode` not ready ❌
  - 🟡  WIP decide how to migrate `Color`+`animations::color_wipe_up`+`animations::color_wipe_down`
  - ⚠️ probably may drop `GenerateComplementaryColor` for simplicity
- [ ] `RandomFillMode` not ready ❌
  - 🟡  WIP decide how to migrate `Color`+`animations::double_side_fill`
  - ⚠️ probably may drop `GenerateRandomColor` for simplicity
- [ ] `PingPongMode` not ready ❌
  - 🟡  WIP decide how to migrate `Color`+`animations::dot_ping_pong`
  - ⚠️ probably may drop `GenerateComplementaryColor` for simplicity
- [ ] `NoiseMode` not ready ❌
   - 🔴 need to clarify how strip buffers may be accessed by user modes
- [ ] `FftMode`  not ready ❌
   - 🔴 need to clarify how microphone FFT may be accessed by user modes
   - 🔴 need to clarify how strip buffers may be accessed by user modes
- [ ] `LiquideMode` not ready ❌
   - 🔴 need to clarify how IMU may be accessed by user modes
   - ⚠️ may drop `GenerateRainbowSwirl` and use static `modes::colors::PaletteTy` color palettes
- [ ] `LiquideRainMode` not ready ❌
   - 🔴 need to clarify how IMU may be accessed by user modes
   - ⚠️ may drop `GenerateRainbowSwirl` and use static `modes::colors::PaletteTy` color palettes

deuxksix, 2025-08-19:
Important side-note:
- all drawing functions / animations speed shall be configured using milliseconds (how much time to completion, etc)
- whenever possible, do not use locally defined ever-increasing counters, prefer only using `ctx.lamp.tick` which is guaranteed to be increased at a constant rate each second (and thus can easily converted to a millisecond amount)

BaptisteHudyma, 2025-12-26:
Resolve in #368

closed 2025-12-26 by BaptisteHudyma